wdoekes/asterisk-chan-dongle #123: E171 no voice

AlexHighTower, 2020-10-27:
Hello,

I have E171 modem with Revision: 21.156.00.00.143
it is recognised by module
dngl-tele2_2 0     Free       9    0    0       TELE2          E171       21.156.00.00.143  xxxxxxxxxxxxxx  xxxxxxxxyyyyyyy  Unknown 

I can make outgoing call via it, BUT, then remove side pick up phone, nothing happens on asterisk side, it is still state "ringing", while remote side have call answered and time is running, but only silence.

If to make incoming call to it - completely nothing on asterisk side... remote side have only ringing at that's all...

voice is enabled
AT^CVOICE?
^CVOICE: 0, 8000, 16, 20

I have similar modem 
dngl-tele2_1 0     Free       13   0    0       TELE2          E173       11.126.85.00.209  yyyyyyyyyyyyyy  xxxxxxxxxxxxxx  Unknown 

and this one works like a charm!

any ideas?

miopa, 2020-11-13:
> dngl-tele2_2 0 Free 9 0 0 TELE2 E171 21.156.00.00.143

E17x with firmware revision 2x... are known to not have voice capability

wdoekes, 2021-10-05:
Closable?

closed 2022-11-04 by wdoekes