wdoekes/asterisk-chan-dongle #126: Unable to send Short SMS

antonatosn, 2020-12-20:
Hi,

I am trying to send Short type SMS but its failing.
When I send the cmd: dongle sms dongle0 1314 YP

I am getting    
-- [dongle0] Error payload: UNKNOWN
 -- [dongle0] Error sending SMS message 0x1f84378

From PDU decoders I saw that there is no SMSC number on PDU.
I have also test a normal number and it works.

Other test:

- I am receiving all the SMS
- Use gammu to send short SMS and its working

Please could you give me an advice?

Thanks

```
OK
]
[2020-12-21 01:23:18] DEBUG[1494]: at_queue.c:126 at_queue_add: [dongle0] insert task with 1 commands begin with 'AT' expected response 'OK' after head of queue
[2020-12-21 01:23:18] DEBUG[1494]: at_queue.c:246 at_queue_run: [dongle0] write command 'AT' expected response 'OK' length 3
]2020-12-21 01:23:18] DEBUG[1494]: at_queue.c:191 at_write: [dongle0] [AT
[2020-12-21 01:23:18] DEBUG[1494]: at_read.c:80 at_read: [dongle0] receive 6 byte, used 6, free 2042, read 0, write 6
[2020-12-21 01:23:18] DEBUG[1494]: at_read.c:95 at_read: [dongle0] [
OK
]
[2020-12-21 01:23:18] DEBUG[1494]: at_response.c:105 at_response_ok: [dongle0] AT sent successfully
[2020-12-21 01:23:18] DEBUG[1494]: at_queue.c:221 at_queue_remove_cmd: [dongle0] remove command 'AT' expected response 'OK' real 'OK' cmd 1/1 flags 0x03 from queue
[2020-12-21 01:23:18] DEBUG[1494]: at_queue.c:69 at_queue_remove: [dongle0] remove task with 1 command(s) begin with 'AT' expected response 'OK' from queue
raspbx*CLI> dongle sms dongle0 1314 YP
[dongle0] SMS queued for send
[2020-12-21 01:23:22] DEBUG[1833]: pdu.c:607 pdu_build: ### dst_toa -> b9  
[2020-12-21 01:23:22] DEBUG[1833]: at_queue.c:126 at_queue_add: [dongle0] insert task with 2 commands begin with 'AT+CMGS' expected response '> ' at tail of queue
[2020-12-21 01:23:22] DEBUG[1833]: at_queue.c:246 at_queue_run: [dongle0] write command 'AT+CMGS' expected response '> ' length 11
]2020-12-21 01:23:22] DEBUG[1833]: at_queue.c:191 at_write: [dongle0] [AT+CMGS=12
[2020-12-21 01:23:22] DEBUG[1494]: at_read.c:80 at_read: [dongle0] receive 4 byte, used 4, free 2044, read 0, write 4
[2020-12-21 01:23:22] DEBUG[1494]: at_read.c:95 at_read: [dongle0] [
> ]
[2020-12-21 01:23:22] DEBUG[1494]: at_queue.c:221 at_queue_remove_cmd: [dongle0] remove command 'AT+CMGS' expected response '> ' real '> ' cmd 1/2 flags 0x00 from queue
[2020-12-21 01:23:22] DEBUG[1494]: at_queue.c:246 at_queue_run: [dongle0] write command 'SMSTEXT' expected response 'OK' length 27
[2020-12-21 01:23:22] DEBUG[1494]: at_queue.c:191 at_write: [dongle0] [00310004B931410000A9025928]
[2020-12-21 01:23:22] DEBUG[1494]: at_read.c:80 at_read: [dongle0] receive 2 byte, used 2, free 2046, read 0, write 2
[2020-12-21 01:23:22] DEBUG[1494]: at_read.c:95 at_read: [dongle0] [
]
OK
]
OK
]
OK
]
[2020-12-21 01:23:52] DEBUG[1494]: at_read.c:80 at_read: [dongle0] receive 19 byte, used 19, free 2029, read 0, write 19
[2020-12-21 01:23:52] DEBUG[1494]: at_read.c:95 at_read: [dongle0] [
+CMS ERROR: 500
]
    -- [dongle0] Error payload: UNKNOWN
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:575 stasis_topic_create_with_detail: Creating topic. name: channel:1608506632.0, detail: 
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:609 stasis_topic_create_with_detail: Topic 'channel:1608506632.0': 0x73b050a8 created
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:575 stasis_topic_create_with_detail: Creating topic. name: cache:6/channel:1608506632.0, detail: 
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:609 stasis_topic_create_with_detail: Topic 'cache:6/channel:1608506632.0': 0x73b059c0 created
[2020-12-21 01:23:52] DEBUG[1494]: channel.c:989 __ast_channel_alloc_ap: Channel 0x73b03880 'Local/report@from-trunk-dongle-00000000;1' allocated
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:575 stasis_topic_create_with_detail: Creating topic. name: channel:1608506632.1, detail: 
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:609 stasis_topic_create_with_detail: Topic 'channel:1608506632.1': 0x73b08928 created
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:575 stasis_topic_create_with_detail: Creating topic. name: cache:7/channel:1608506632.1, detail: 
[2020-12-21 01:23:52] DEBUG[1494]: stasis.c:609 stasis_topic_create_with_detail: Topic 'cache:7/channel:1608506632.1': 0x73b09240 created
[2020-12-21 01:23:52] DEBUG[1494]: channel.c:989 __ast_channel_alloc_ap: Channel 0x73b07620 'Local/report@from-trunk-dongle-00000000;2' allocated
[... 22 lines cut ...]
[2020-12-21 01:23:52] DEBUG[1497]: app_queue.c:2589 device_state_cb: Device 'Local/report@from-trunk-dongle' changed to state '4' (Invalid) but we don't care because they're not a member of any queue.
[2020-12-21 01:23:52] DEBUG[1429]: cdr.c:1478 cdr_object_finalize: Finalized CDR for Local/report@from-trunk-dongle-00000000;1 - start 1608506632.332965 answer 0.000000 end 1608506632.335885 dur 0.002 bill 1608506632.335 dispo FAILED
[2020-12-21 01:23:52] DEBUG[1429]: cdr.c:3541 post_cdr: Skipping CDR for Local/report@from-trunk-dongle-00000000;1 since we weren't answered
[2020-12-21 01:23:52] DEBUG[1836][C-00000001]: stasis.c:438 topic_dtor: Destroying topic. name: cache:6/channel:1608506632.0, detail: 
[2020-12-21 01:23:52] DEBUG[1836][C-00000001]: stasis.c:446 topic_dtor: Topic 'cache:6/channel:1608506632.0': 0x73b059c0 destroyed
[2020-12-21 01:23:52] DEBUG[1836][C-00000001]: stasis.c:438 topic_dtor: Destroying topic. name: channel:1608506632.0, detail: 
[2020-12-21 01:23:52] DEBUG[1836][C-00000001]: stasis.c:446 topic_dtor: Topic 'channel:1608506632.0': 0x73b050a8 destroyed
[2020-12-21 01:23:52] DEBUG[1412]: devicestate.c:361 _ast_device_state: No provider found, checking channel drivers for Local - report@from-trunk-dongle
[2020-12-21 01:23:52] DEBUG[1412]: core_local.c:353 local_devicestate: Checking if extension report@from-trunk-dongle exists (devicestate)
[2020-12-21 01:23:52] DEBUG[1412]: devicestate.c:466 do_state_change: Changing state for Local/report@from-trunk-dongle - state 4 (Invalid)
[2020-12-21 01:23:52] DEBUG[1497]: app_queue.c:2589 device_state_cb: Device 'Local/report@from-trunk-dongle' changed to state '4' (Invalid) but we don't care because they're not a member of any queue.
OK
]
[2020-12-21 01:24:01] DEBUG[1845]: manager.c:6688 process_message: Running action 'Login'
[2020-12-21 01:24:02] DEBUG[1494]: at_queue.c:126 at_queue_add: [dongle0] insert task with 1 commands begin with 'AT' expected response 'OK' after head of queue
[2020-12-21 01:24:02] DEBUG[1494]: at_queue.c:246 at_queue_run: [dongle0] write command 'AT' expected response 'OK' length 3
]2020-12-21 01:24:02] DEBUG[1494]: at_queue.c:191 at_write: [dongle0] [AT
[2020-12-21 01:24:02] DEBUG[1494]: at_read.c:80 at_read: [dongle0] receive 6 byte, used 6, free 2042, read 0, write 6
[2020-12-21 01:24:02] DEBUG[1494]: at_read.c:95 at_read: [dongle0] [
OK
]
[2020-12-21 01:24:02] DEBUG[1494]: at_response.c:105 at_response_ok: [dongle0] AT sent successfully
[2020-12-21 01:24:02] DEBUG[1494]: at_queue.c:221 at_queue_remove_cmd: [dongle0] remove command 'AT' expected response 'OK' real 'OK' cmd 1/1 flags 0x03 from queue
[2020-12-21 01:24:02] DEBUG[1494]: at_queue.c:69 at_queue_remove: [dongle0] remove task with 1 command(s) begin with 'AT' expected response 'OK' from queue
```

archekb, 2021-02-02:
https://github.com/wdoekes/asterisk-chan-dongle/issues/84#issuecomment-769634173

archekb, 2021-02-02:
#84

archekb, 2021-02-02:
the same problem https://github.com/wdoekes/asterisk-chan-dongle/issues/84#issuecomment-769634173

standardgbg, 2022-11-20:
Hi.

Any news on this? I Sweden the operators are using a non-standard mobile number for the users to response to.
But when i try to send to the number 71550 I get the following:
```
asterisk*CLI>dongle sms dongle0 71550 CKSTOP
[dongle0] SMS queued for send
[Nov 20 14:28:58]     -- [dongle0] Error payload: UNKNOWN
[Nov 20 14:28:58]     -- [dongle0] Error sending SMS message 0x7fd82c000b00
[Nov 20 14:28:58] ERROR[30808]: at_response.c:287 log_cmd_response_error: [dongle0] Error sending SMS message 0x7fd82c000b00 S                                                                                        MSTEXT
[Nov 20 14:28:58] WARNING[31628][C-00000056]: pbx.c:4461 __ast_pbx_run: Channel 'Local/report@from-trunk-dongle-00000033;1' sent to invalid extension but no invalid handler: context,exten,priority=from-trunk-dongle,report,1

```

wdoekes, 2022-11-21:
@standardgbg : Do you have debug logs, like the OP?

Especially this part:
```
[2020-12-21 01:23:22] DEBUG[1494]: at_queue.c:191 at_write: [dongle0] [00310004B931410000A9025928�]
```
That garbage character at the end looks suspicious for instance. The rest _seems_ legit according to https://www.diafaan.com/sms-tutorials/gsm-modem-tutorial/online-sms-pdu-decoder/

@antonatosn: do you have the PDU that Gammu generates?

wdoekes, 2022-11-21:
@standardgbg : You could try fiddling with the number-type.

Till exempel:
```diff
diff --git a/pdu.c b/pdu.c
index 94f7427..a6f7b6e 100644
--- a/pdu.c
+++ b/pdu.c
@@ -212,8 +212,7 @@
 #define NUMBER_TYPE_INTERNATIONAL	(TP_A_EXT_NOEXT | TP_A_TON_INTERNATIONAL | TP_A_NPI_TEL_E164_E163) /* 0x91 */
 #define NUMBER_TYPE_NATIONAL		(TP_A_EXT_NOEXT | TP_A_TON_SUBSCRIBERNUM | TP_A_NPI_NATIONALNUM) /* 0xC8 */
 #define NUMBER_TYPE_ALPHANUMERIC	(TP_A_EXT_NOEXT | TP_A_TON_ALPHANUMERIC | TP_A_NPI_UNKNOWN) /* 0xD0 */
-/* maybe NUMBER_TYPE_NETWORKSHORT should be 0xB1 ??? */
-#define NUMBER_TYPE_NETWORKSHORT	(TP_A_EXT_NOEXT | TP_A_TON_NETSPECIFIC | TP_A_NPI_PRIVATENUM) /* 0xB9 */
+#define NUMBER_TYPE_NETWORKSHORT	(TP_A_EXT_NOEXT | TP_A_TON_NETSPECIFIC | TP_A_NPI_TEL_E164_E163) /* 0xB1 */
 #define NUMBER_TYPE_UNKNOWN		(TP_A_EXT_NOEXT | TP_A_TON_UNKNOWN | TP_A_NPI_TEL_E164_E163) /* 0x81 */
 
 /* Reject Duplicate */
```
See https://en.wikipedia.org/wiki/GSM_03.40 for possible combinations.

Let us know if changing the value helps.

antonatosn, 2022-11-21:
Hey folks,

I solve it in a different way. I am sending it as a PDU command on asterisk -x.

/usr/sbin/asterisk -rx "dongle pdu dongle0 00210005819118F000081003a70391039c039f03930395039b039f"
This is an example of a 5 digit SMS (charity) in Greek language.

If I recall, the problem was on SMSC. Since the implementation was not sending SMSC each time but relies on SIM to append it later (which for some doesn't do it).

standardgbg, 2022-12-28:
Hi.

Can someone help me figure out why I'm not able to send pdu?

My avalible commands:
dongle cmd                     -- Send commands to port for debugging
dongle discovery               -- Discovery devices and create config
dongle reload                  -- Reload dongle
dongle remove                  -- Remove dongle
dongle reset                   -- Reset dongle now
dongle restart                 -- Restart dongle
dongle show device settings    -- Show Dongle device settings
dongle show device state       -- Show Dongle device state
dongle show device statistics  -- Show Dongle device statistics
dongle show devices            -- Show Dongle devices state
dongle show version            -- Show module version
dongle sms                     -- Send SMS from the dongle
dongle start                   -- Start dongle
dongle stop                    -- Stop dongle
dongle ussd                    -- Send USSD commands to the dongle

standardgbg, 2022-12-28:
> @standardgbg : Do you have debug logs, like the OP?
>
> Especially this part:
>
> ```
> [2020-12-21 01:23:22] DEBUG[1494]: at_queue.c:191 at_write: [dongle0] [00310004B931410000A9025928�]
> ```
>
> That garbage [...]

dongle sms dongle0 71550 CKSTOP
[dongle0] SMS queued for send
[Dec 28 23:24:43] DEBUG[25124]: at_queue.c:126 at_queue_add: [dongle0] insert task with 2 commands begin with 'AT+CMGS' expected response '> ' at tail of queue
[Dec 28 23:24:43] DEBUG[25124]: at_queue.c:246 at_queue_run: [dongle0] write command 'AT+CMGS' expected response '> ' length 11
]Dec 28 23:24:43] DEBUG[25124]: at_queue.c:191 at_write: [dongle0] [AT+CMGS=17

[Dec 28 23:24:43] DEBUG[24358]: at_read.c:80 at_read: [dongle0] receive 4 byte, used 4, free 2044, read 0, write 4
[Dec 28 23:24:43] DEBUG[24358]: at_read.c:95 at_read: [dongle0] [
> ]
[Dec 28 23:24:43] DEBUG[24358]: at_queue.c:221 at_queue_remove_cmd: [dongle0] remove command 'AT+CMGS' expected response '> ' real '> ' cmd 1/2 flags 0x00 from queue
[Dec 28 23:24:43] DEBUG[24358]: at_queue.c:246 at_queue_run: [dongle0] write command 'SMSTEXT' expected response 'OK' length 37
[Dec 28 23:24:43] DEBUG[24358]: at_queue.c:191 at_write: [dongle0] [00310005B91755F00000A906C3E594FA8402]
[Dec 28 23:24:43] DEBUG[24358]: at_read.c:80 at_read: [dongle0] receive 2 byte, used 2, free 2046, read 0, write 2
[Dec 28 23:24:43] DEBUG[24358]: at_read.c:95 at_read: [dongle0] [
]
[Dec 28 23:24:47] DEBUG[19348]: at_queue.c:126 at_queue_add: [dongle1] insert task with 1 commands begin with 'AT' expected response 'OK' after head of queue
[Dec 28 23:24:47] DEBUG[19348]: at_queue.c:246 at_queue_run: [dongle1] write command 'AT' expected response 'OK' length 3
]Dec 28 23:24:47] DEBUG[19348]: at_queue.c:191 at_write: [dongle1] [AT
[Dec 28 23:24:47] DEBUG[19348]: at_read.c:80 at_read: [dongle1] receive 6 byte, used 6, free 2042, read 0, write 6
[Dec 28 23:24:47] DEBUG[19348]: at_read.c:95 at_read: [dongle1] [
OK
]
[Dec 28 23:24:47] DEBUG[19348]: at_response.c:105 at_response_ok: [dongle1] AT sent successfully
[Dec 28 23:24:47] DEBUG[19348]: at_queue.c:221 at_queue_remove_cmd: [dongle1] remove command 'AT' expected response 'OK' real 'OK' cmd 1/1 flags 0x03 from queue
[Dec 28 23:24:47] DEBUG[19348]: at_queue.c:69 at_queue_remove: [dongle1] remove task with 1 command(s) begin with 'AT' expected response 'OK' from queue
[Dec 28 23:24:50] DEBUG[15167]: res_pjsip_registrar.c:1225 check_expiration_thread: Woke up at 1672266290  Interval: 30
[Dec 28 23:24:50] DEBUG[15167]: res_pjsip_registrar.c:1232 check_expiration_thread: Expiring 0 contacts
[Dec 28 23:24:50] DEBUG[24358]: at_read.c:80 at_read: [dongle0] receive 27 byte, used 27, free 2021, read 0, write 27
[Dec 28 23:24:50] DEBUG[24358]: at_read.c:95 at_read: [dongle0] [
^BOOT:34892182,0,0,0,20
]
[Dec 28 23:24:50] DEBUG[24358]: at_response.c:1978 at_response: [dongle0] Ignoring unknown result: ''
[Dec 28 23:24:57] DEBUG[19348]: at_queue.c:126 at_queue_add: [dongle1] insert task with 1 commands begin with 'AT' expected response 'OK' after head of queue
[Dec 28 23:24:57] DEBUG[19348]: at_queue.c:246 at_queue_run: [dongle1] write command 'AT' expected response 'OK' length 3
]Dec 28 23:24:57] DEBUG[19348]: at_queue.c:191 at_write: [dongle1] [AT
[Dec 28 23:24:58] DEBUG[19348]: at_read.c:80 at_read: [dongle1] receive 6 byte, used 6, free 2042, read 0, write 6
[Dec 28 23:24:58] DEBUG[19348]: at_read.c:95 at_read: [dongle1] [
OK
]
[Dec 28 23:24:58] DEBUG[19348]: at_response.c:105 at_response_ok: [dongle1] AT sent successfully
[Dec 28 23:24:58] DEBUG[19348]: at_queue.c:221 at_queue_remove_cmd: [dongle1] remove command 'AT' expected response 'OK' real 'OK' cmd 1/1 flags 0x03 from queue
[Dec 28 23:24:58] DEBUG[19348]: at_queue.c:69 at_queue_remove: [dongle1] remove task with 1 command(s) begin with 'AT' expected response 'OK' from queue
[Dec 28 23:25:07] DEBUG[19348]: at_read.c:80 at_read: [dongle1] receive 27 byte, used 27, free 2021, read 0, write 27
[Dec 28 23:25:07] DEBUG[19348]: at_read.c:95 at_read: [dongle1] [
^BOOT:41830149,0,0,0,20
]
[Dec 28 23:25:08] DEBUG[24358]: at_read.c:80 at_read: [dongle0] receive 19 byte, used 20, free 2028, read 26, write 46
[Dec 28 23:25:08] DEBUG[24358]: at_read.c:95 at_read: [dongle0] [

+CMS ERROR: 500
]
[Dec 28 23:25:08] DEBUG[24358]: at_read.c:131 at_read_result_iov: [dongle0] multiline response
[Dec 28 23:25:08]     -- [dongle0] Error payload: UNKNOWN
[Dec 28 23:25:08] DEBUG[24358]: channel.c:989 __ast_channel_alloc_ap: Channel 0x7fc224580480 'Local/report@from-trunk-dongle-00000007;1' allocated
[Dec 28 23:25:08] DEBUG[24358]: channel.c:989 __ast_channel_alloc_ap: Channel 0x7fc224583b10 'Local/report@from-trunk-dongle-00000007;2' allocated
[Dec 28 23:25:08] DEBUG[24358]: channel.c:1229 set_channel_vars: [dongle0] Setting chanvar DONGLENAME = dongle0
[Dec 28 23:25:08] DEBUG[24358]: channel.c:1229 set_channel_vars: [dongle0] Setting chanvar DONGLEPROVIDER = Tele 2 AB
[Dec 28 23:25:08] DEBUG[24358]: channel.c:1229 set_channel_vars: [dongle0] Setting chanvar DONGLEIMEI = REMOVEDBYME :)
[Dec 28 23:25:08] DEBUG[24358]: channel.c:1229 set_channel_vars: [dongle0] Setting chanvar DONGLEIMSI = REMOVEDBYME :)
[Dec 28 23:25:08] DEBUG[24358]: channel.c:1229 set_channel_vars: [dongle0] Setting chanvar DONGLENUMBER = Unknown
[Dec 28 23:25:08]     -- [dongle0] Error sending SMS message 0x7fc22800ae20
[... cut ...]
[Dec 28 23:25:08] DEBUG[15139]: devicestate.c:361 _ast_device_state: No provider found, checking channel drivers for Local - report@from-trunk-dongle
[Dec 28 23:25:08] DEBUG[15139]: core_local.c:348 local_devicestate: Checking if extension report@from-trunk-dongle exists (devicestate)
[Dec 28 23:25:08] DEBUG[15139]: devicestate.c:466 do_state_change: Changing state for Local/report@from-trunk-dongle - state 4 (Invalid)
[Dec 28 23:25:08] DEBUG[15207]: app_queue.c:2585 device_state_cb: Device 'Local/report@from-trunk-dongle' changed to state '4' (Invalid) but we don't care because they're not a member of any queue.

antonatosn, 2022-12-29:
@standardgbg dongle pdu has been removed on: 

https://github.com/wdoekes/asterisk-chan-dongle/commit/868bbd41fbf2911cdfba483e50c8e99a79d84258

Currently I am using raspbx which uses the bg111 version.

antonatosn, 2023-05-07:
I try some changes on PDU.c

the `#define NUMBER_TYPE_NETWORKSHORT` should be 
`#define NUMBER_TYPE_NETWORKSHORT 0x81`

also about the garbage char on [comment](https://github.com/wdoekes/asterisk-chan-dongle/issues/126#issuecomment-1366961276) it's not garbage at all. It is the CTRL+Z or CHR(26)

Despite that SMS is working now. I suggest we should brink back the cli command to send SMS in PDU form.

AlexanderZhirov, 2023-05-26:
Unable to send message

```sh
solus*CLI> dongle sms dongle0 +79XXXXXXXXX "This is big sms test"
[dongle0] SMS queued for send
    -- [dongle0] Error payload: UNKNOWN
    -- [dongle0] Error sending SMS message 0x7f057c050b50
[May 26 19:24:46] ERROR[57]: at_response.c:287 log_cmd_response_error: [dongle0] Error sending SMS message 0x7f057c050b50 SMSTEXT
[May 26 19:24:46] WARNING[79][C-00000005]: pbx.c:4507 __ast_pbx_run: Channel 'Local/report@incoming-00000003;1' sent to invalid extension but no invalid handler: context,exten,priority=incoming,report,1
```

antonatosn, 2023-05-26:
> Unable to send message
> 
> ```shell
> solus*CLI> dongle sms dongle0 +79XXXXXXXXX "This is big sms test"
> [dongle0] SMS queued for send
>     -- [dongle0] Error payload: UNKNOWN
>     -- [dongle0] Error sending SMS message 0x7f057c050b50
> [May 26 19:24:46] ERROR[57]: at_response.c:287 log_cmd_response_error: [dongle0] Error sending SMS message 0x7f057c050b50 SMSTEXT
> [May 26 19:24:46] WARNING[79][C-00000005]: pbx.c:4507 __ast_pbx_run: Channel 'Local/report@incoming-00000003;1' sent to invalid extension but no invalid handler: context,exten,priority=incoming,report,1
> ```
> 
> P.S. Sorry, just don't have any money on my balance sheet 😄

Please, enable the debug to see the AT commands and responses.

AlexanderZhirov, 2023-05-26:
> Please, enable the debug to see the AT commands and responses.

I didn't have enough funds to send SMS.

```
sent to invalid extension but no invalid handler: context,exten,priority=incoming,report,1
```

Is there any variable containing the error text to handle it in exten `report`?

antonatosn, 2023-05-26:
> > Please, enable the debug to see the AT commands and responses.
> 
> I didn't have enough funds to send SMS.
> 
> ```
> sent to invalid extension but no invalid handler: context,exten,priority=incoming,report,1
> ```
> 
> Is there any variable containing the error text to handle it in exten `report`?

Unfortunately, I don't know but nice to have it.