wdoekes/asterisk-chan-dongle #129: One-way audio with E173 device

muks, 2021-01-16:
Versions:

* Asterisk version asterisk-17.9.0-1.fc33.x86_64 on Fedora 33
* chan_dongle master branch at commit 2bd3fc2cc31562826c8341c5584fe26def9ca808 (current as of today).

With a Huawei E173 device:

* Audio from GSM network via chan_dongle to PJSIP client is heard correctly on the SIP client
* Audio from PJSIP client to GSM network via chan_dongle is not heard on the other party's phone

When calling into an IVR extension from GSM the network extension, the IVR playback audio is heard on the calling party's phone correctly. It's when dialling out from Asterisk to the GSM network that the problem seems to happen.

2-way audio used to work before in an older version of chan_dongle (but I don't recall how old).

muks, 2021-01-16:
I keenly want this to work again. If you want me to supply any logs, or test any patches with further debug logging, please let me know and I'll do it.

muks, 2021-01-17:
It turned out to be a firewall problem on the new Asterisk machine which was blocking media. The problem had nothing to do with chan_dongle. I'm profusely sorry for this.

Closing bug.

closed 2021-01-17 by muks

wdoekes, 2021-01-17:
No worries. Thanks for closing :)

vogelfreiheit, 2021-12-17:
I'm experiencing this and it is within the same VLAN and no firewall configured on the Asterisk server:

`
[2021-12-17 19:07:51] NOTICE[19243]: res_pjsip_sdp_rtp.c:150 rtp_check_timeout: Disconnecting channel 'PJSIP/888-00000002' for lack of audio RTP activity in 30 seconds`

Audio path from GSM->SIP works fine. But the other party cannot hear the audio from the SIP->GSM side.