wdoekes/asterisk-chan-dongle #132: E1550: outbound sound issues

rusangl, 2021-01-30:
I use RasPBX (raspberrypi 3b, debian stretch, Freepbx 15.0.17.9, asterisk 16.15.0, Huawei e1550 with firmware 11.608.12.00.143).
Inbound calls work perfectly, but when I call from SIP to GSM the GSM-party doesn’t hear the opposite side for about 10-12 seconds. In the logs I see nothing special except something like NOTICE: translate.c:597 ast_translate: 492 lost frame(s) 493/0 (slin@8000)->(gsm@8000).
No matter what codec I use.
I would really appreciate it if anybody could explain what does this mean and how to solve it?

archekb, 2021-02-02:
Try update your dongle firmware to 11.609.20.00.00.B418 it's 100% works firmware (dongle can lost your sim card, if sim is too old < 2015). I have asterisk 18.1.1 and 2 e1550 dongles with 11.609.20.00.00.B418 and it works fine.

rusangl, 2021-02-02:
@archekb , thank you much for the answer, I'll try to update firmware tomorrow. Do you also use raspberry (or any other arm), or an x86 PC?

Some people in the Internet wrote that my firmware is OK, but for x86 and older Asterisk versions

archekb, 2021-02-02:
I use asterisk in docker (alpine:latest) on ubuntu 20.10 server amd64 (x86).

wdoekes, 2021-10-05:
All good?

archekb, 2021-10-05:
After that I was try

rusangl, 2021-10-09:
Resently I've found that the problem depends on the sip-client. 
Usually I use GS Wave and there was a silence as I described in the first post.
I tried Sipnetic - warnings continued, but GSM party can hear voice from the beginning.