wdoekes/asterisk-chan-dongle #133: SMS in UNICODE are dropped by dongle module

hellkaim, 2021-02-10:
Hi.
I have a system with Asterisk and your chan_dongle module running under Debian. THANK YOU for maintaining it for us!

However I experience a strange issue with Asterisk 16.16.0 (and 18.X but that was expected) and your current chan_dongle:
Outgoing SMS with UNICODE encoding (non-English text) are dropped by the dongle_module.

The issue is only reproducible with Asterisk 16.X branch. Under Asterisk 13.38.1 all SMS's can be send without any issues.
The incomming unicod SMSs are delivered also with no issue. The bug is not related to the dongle HW or FW as I have E1750 (11.126.13.00.00) and E1550 (11.608.14.15.311) modems and they all fail the same.

The exact error is:
`Executing [+1234567890@outgoing-sms:13] DongleSendSMS("Message/ast_msg_queue", "dongle0,+1234567890,"могпгпшппшршрш",1140,yes") in new stack
-- Executing [+1234567890@outgoing-sms:14] Hangup("Message/ast_msg_queue", "") in new stack
== Spawn extension (outgoing-sms, +1234567890, 14) exited non-zero on 'Message/ast_msg_queue'
-- [dongle0] Error sending SMS message 0xffff640074b0
[Feb 10 07:59:08] ERROR[26729]: at_response.c:500 at_response_error: [dongle0] Error sending SMS message 0xffff640074b0`

miopa, 2021-02-11:
I tried this from bash on my Asterisk 16.16 and it worked:

`asterisk -rx 'dongle sms dongle0 +1234567890 могпгпшппшршрш'`

I'll let you know when I test from dialplan

hellkaim, 2021-02-12:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEUAAABFCAYAAAAcjSspAAAABHNCSVQICAgIfAhkiAAAAClJREFUeJztwYEAAAAAw6D5U1/gCFUBAAAAAAAAAAAAAAAAAAAAAADAMUqpAAFjVpDmAAAAAElFTkSuQmCC###24x24:true###][Tracking image][https://github.com/notifications/beacon/AJTQCS2JNT27CK7GG3YFFMTS6Q6OXA5CNFSM4XNFGKO2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFZN6YII.gif]

Thanks. We had this issue only with outgoing messages coming through pjsip trunk.

miopa, 2021-11-01:
I tried similar thing on Asterisk 16.20 and latest chan_dongle, and SMS was properly sent

```
-- Executing [sms@from-trunk:5] DongleSendSMS("Local/sms@from-trunk-000001d5;1", "dongle0,+1234567890,"тест кирилица",1440,1,UNKNOWN") in new stack
```

wdoekes, 2022-11-04:
Can't tell what's wrong. Is anything wrong?