wdoekes/asterisk-chan-dongle #134: E1550 chan_dongle broke ppp

AlexHighTower, 2021-02-19:
Hello

I have couple of E1550 modem
Before now both of them was used like this:
- first usb was used for PPP client to connect to internet
- second usb port was used with self written application to send\receive sms
- voice was not used

Today I decided to use asterisk to send\receive sms and as bonus get voice calls and now I have problems...

Main problem is that PPP now don't work!
connection log looks like this:

> Feb 19 21:13:25 home chat[11091]: abort on (BUSY)
> Feb 19 21:13:25 home chat[11091]: abort on (VOICE)
> Feb 19 21:13:25 home chat[11091]: abort on (NO CARRIER)
> Feb 19 21:13:25 home chat[11091]: abort on (NO DIALTONE)
> Feb 19 21:13:25 home chat[11091]: abort on (NO DIAL TONE)
> Feb 19 21:13:25 home chat[11091]: abort on (NO ANSWER)
> Feb 19 21:13:25 home chat[11091]: abort on (DELAYED)
> Feb 19 21:13:25 home chat[11091]: abort on (ERROR)
> Feb 19 21:13:25 home chat[11091]: send (AT^M)
> Feb 19 21:13:25 home chat[11091]: timeout set to 15 seconds
> Feb 19 21:13:25 home chat[11091]: expect (OK)
> Feb 19 21:13:25 home chat[11091]: AT^M^M
> Feb 19 21:13:25 home chat[11091]: OK
> Feb 19 21:13:25 home chat[11091]:  -- got it
> Feb 19 21:13:25 home chat[11091]: send (ATE1^M)
> Feb 19 21:13:25 home chat[11091]: expect (OK)
> Feb 19 21:13:25 home chat[11091]: ^M
> Feb 19 21:13:25 home chat[11091]: ATE1^M^M
> Feb 19 21:13:25 home chat[11091]: OK
> Feb 19 21:13:25 home chat[11091]:  -- got it
> Feb 19 21:13:25 home chat[11091]: send (AT^M)
> Feb 19 21:13:25 home chat[11091]: expect (OK)
> Feb 19 21:13:25 home chat[11091]: ^M
> Feb 19 21:13:25 home chat[11091]: AT^M^M
> Feb 19 21:13:25 home chat[11091]: OK
> Feb 19 21:13:25 home chat[11091]:  -- got it
> Feb 19 21:13:25 home chat[11091]: send (AT+CGDCONT=1,"IP","internet"^M)
> Feb 19 21:13:26 home chat[11091]: expect (OK)
> Feb 19 21:13:26 home chat[11091]: ^M
> Feb 19 21:13:26 home chat[11091]: AT+CGDCONT=1,"IP","internet"^M^M
> Feb 19 21:13:26 home chat[11091]: OK
> Feb 19 21:13:26 home chat[11091]:  -- got it
> Feb 19 21:13:26 home chat[11091]: send (ATD*99***1#^M)
> Feb 19 21:13:26 home chat[11091]: timeout set to 22 seconds
> Feb 19 21:13:26 home chat[11091]: expect (CONNECT)
> Feb 19 21:13:26 home chat[11091]: ^M
> Feb 19 21:13:26 home chat[11091]: ATD*99***1#^M^M
> Feb 19 21:13:26 home chat[11091]: CONNECT
> Feb 19 21:13:26 home chat[11091]:  -- got it
> Feb 19 21:13:26 home chat[11091]: send (^M)
> Feb 19 21:13:26 home pppd[10016]: Serial connection established.
> Feb 19 21:13:26 home pppd[10016]: Using interface ppp99
> Feb 19 21:13:26 home pppd[10016]: Connect: ppp99 <--> /dev/ttyUSB1
> Feb 19 21:13:27 home pppd[10016]: PAP authentication succeeded
> Feb 19 21:13:27 home pppd[10016]: Hangup (SIGHUP)
> Feb 19 21:13:27 home pppd[10016]: Modem hangup
> Feb 19 21:13:27 home pppd[10016]: Connection terminated.

as you see - immediate hangup

time to time log looks like

> Feb 19 21:23:59 home chat[12958]: abort on (BUSY)
> Feb 19 21:23:59 home chat[12958]: abort on (VOICE)
> Feb 19 21:23:59 home chat[12958]: abort on (NO CARRIER)
> Feb 19 21:23:59 home chat[12958]: abort on (NO DIALTONE)
> Feb 19 21:23:59 home chat[12958]: abort on (NO DIAL TONE)
> Feb 19 21:23:59 home chat[12958]: abort on (NO ANSWER)
> Feb 19 21:23:59 home chat[12958]: abort on (DELAYED)
> Feb 19 21:23:59 home chat[12958]: abort on (ERROR)
> Feb 19 21:23:59 home chat[12958]: send (AT^M)
> Feb 19 21:23:59 home chat[12958]: timeout set to 15 seconds
> Feb 19 21:23:59 home chat[12958]: expect (OK)
> Feb 19 21:23:59 home chat[12958]: ^M
> Feb 19 21:23:59 home chat[12958]: NO CARRIER
> Feb 19 21:23:59 home chat[12958]:  -- failed
> Feb 19 21:23:59 home chat[12958]: Failed (NO CARRIER)
> Feb 19 21:23:59 home pppd[10016]: Connect script failed

second problem is that dongle initialise very long in in asterisk... 
state shows "GSM not re"
after that for several seconds it shows WRONG provider name! 

but main problem, that even if I remove working with dongle from asterisk, REPLUG it, PPP still not work (((

can someone tell me is chan_dongle code sends some AT command which can broke PPP connection and how to restore it?

AlexHighTower, 2021-03-01:
update from my side...
I've made small investigation with 5 modems E1550 I have
originally that was branded to different providers
all of them was:
- unlocked
- voice enabled (using E1550_Russia_MTS_PatchDataCard1.0.0.3.exe file)
- changed imei (using imei-recovery-template-huawei-e1550-100000000000009.exe file)

only one of them can have ppp working while initialised by chan_dongle, rest four fails as I wrote before.

but there is no problem if I use own software to work with sms...

I decided to dump all AT command chan_dongle sends to modem and reproduce it manually via minicom

I started ping command and send all command one-by-one...

and found command which breaks... it is _AT+COPS=0,0_

so,  I remove this command from at_command.c, recompile and now I can use ppp and voice and the time!!!

the only problem that now instead of operator text can its numerical value is shown...

but for me it is not a problem because in any case it was shown wrong... I use sim cards of virtual operator (MVNO) and instead of name of virtual operator it was shown name of "base" operator...

it is possible to create setting for this case? so by default use AT+COPS=0,0 and special setting to disable it?
and may be there is another way to get operator name?

wdoekes, 2021-10-05:
Is this still an issue?

AlexHighTower, 2021-10-06:
yes, it is still an issue for me
tested on  3d046f7 

as soon as while dongle initialization command AT+COPS=0,0 then ppp failed to connect
I recompile 3d046f7
with commented line `ATQ_CMD_DECLARE_ST(CMD_AT_COPS_INIT, "AT+COPS=0,0\r"),`
and now ppp works and dongle initialized successfully. the only issue here that instead on operator name I see its mcc mnc (numbers), but I don't care actually

wdoekes, 2021-10-06:
And if you replace `ATQ_CMD_DECLARE_ST` with `ATQ_CMD_DECLARE_STI` ?

That should make it ignore a failed `AT+COPS=0,0` command (basically the same fix as 3d046f7).

AlexHighTower, 2021-10-06:
the problem it that after executing AT+COPS=0,0 on dongle it is not possible to set up ppp session...
response for AT+COPS=0,0 is OK
it looks very strange and I can't find explanation, but for me it is 100% reproducible, see my first message with ppp connection log
and it is not chan_dongle bug, is it "bug" (or whatever) of dongle itself

I disable chan_dongle
plug dongle
start up ppp session
works fine

start minicom on second at port and send AT+COPS=0,0
ppp session drops and unable to establish
only way to fix it - replug dongle or sent at+cfun=1,1

may be it is only my only problem with particular dongle fw and the only that can be implemented in chan_dongle - kind of switch to disable getting operator name...

I think issue can be closed for now..

wdoekes, 2021-11-03:
Closing per reporters request. I have no idea what to do here.

closed 2021-11-03 by wdoekes