wdoekes/asterisk-chan-dongle #137: Configuring a 4G Module

kidmock, 2021-03-14:
Since appears this is the active fork I'll paste my question here:

I have a SIM7600 module that I used primarily for SMS messages from my home automation and alarm system.

For my original use case it works great. However, I figure I should be able to use it for my home phone as well.

I can plug a headset in to the audio jack and use the AT commands to make and receive phone calls, no problem.

I stood up a quick and dirty asterisk box on a raspberry pi and configured chan_dongle. This doesn't seem to work.
chan_dongle sees:
```
Voice : No
SMS : Yes
```
Using chan_dongle as a trunk I can make calls, but have no audio in either direction. I can get an extension to ring, but again I have no audio in either direction.

It would seem possible that I should be able to make this work but can't seem to find information I'm looking for.

My question is does anyone know of a way in which I can use the audio jack in combination with a way to issue AT commands to this module.

Another person asked how I configured the module which is:
I have the sim7600g-h 4g which is the International version. I'm Using a T-Mobile MNVO in the US that provides Voice and Text only.

I have the jumpers set to position C and set the power on boot by placing a jumper between 3.3V and GND.

I plug the module in the the USB port (not the USB to UART) and my linux hosts recognize/create the following devices

```
/dev/ttyUSB0  - Diagnostic Port
/dev/ttyUSB1  - NEMA GPS 
/dev/ttyUSB2 - Console port for AT Command Port
/dev/ttyUSB3 - Console port for AT Command Port
/dev/ttyUSB4 - Audio Port
/dev/cdc-wdm0 - Network Interface
```

I set the carrier provided APN through a serial terminal emulator (in my case minicom)

```
AT+CGDCONT=1,"IPV4V6","APN_1"
AT+CGDCONT=2,"IPV4V6","APN_2"
```

For my original use case to send text alerts from my security and and home automation system.
I would pipe it to /dev/ttyUSB3 which is something like:

```
echo  "AT+CMGS=\"+15551231234\""> /dev/ttyUSB3
echo "My Message\x1a" >  /dev/ttyUSB3
```

Since I began playing with asterisk, I configured chan_dongle as:

```
[dongle0]
data=/dev/ttyUSB2
audio=/dev/ttyUSB4
```

I've tried using different ports for audio none of them work

but sending SMS messages is as easy as:

```
asterisk -rx "dongle sms dongle0 +15551231234 My Message" 
```

Like I said above, I can plug a headset into the 3.5 mm jack and I can get audio. I just can seem to route the audio into asterisk even though I can get an extension to "ring"

spacedream, 2021-04-09:
If you looking for help you need provide full exact name of module (SIM7600 have few different ones) and better find pdf documentation on AT commands to this module.

spi43984, 2021-04-20:
Check [https://github.com/bg111/asterisk-chan-dongle/issues/276#fromHistory](url)

I see the benefit of the SIM7600as well as it is available and doesn't need any firmware modifications as many Huawei UMTS dongles need to support audio.

I played around a little bit with a SIM7600E and figured out that chan_dongle uses AT^DDSETEX to enable PCM on the ttyUSB interface as soon the audio call is started. SIM7600 expects AT^CPCMREG instead.

So there are two ways to solve this:
1) change chan_dongle and recompile it yourself - the command you'll find in [https://github.com/wdoekes/asterisk-chan-dongle/blob/master/at_command.h](url) 

2) wait for a firmware update for the SIM7600 which supports AT^DDSETEX as well

The former you can check out right now, the latter has the beauty that both UMTS dongles and SIM7600 would be supported.

BTW - connecting multiple SIM7600 with a rpi via USB should work - has anyone tried that yet? As the rpi has only one UART stacking them on GPIO pins might work if the're all additionally connected via USB but I haven't tried that yet.

spi43984, 2021-04-23:
> BTW - connecting multiple SIM7600 with a rpi via USB should work - has anyone tried that yet? As the rpi has only one UART stacking them on GPIO pins might work if the're all additionally connected via USB but I haven't tried that yet.

I talked to Waveshare who I got my SIM7600E-H from - they confirmed that it is possible to connect two SIM7600 to one rpi.

wdoekes, 2022-01-17:
Please turn this into a PR if you want any fixes merged. I don't see anything I can do with this right now.

closed 2022-01-17 by wdoekes