wdoekes/asterisk-chan-dongle #142: Huawei E367 is not detected by IMEI/IMSI, only by data/Audio port exactly specified !

CpServiceSpb, 2021-05-01:
Almost the same issue as [https://github.com/bg111/asterisk-chan-dongle/issues/266](Huawei K3765 not detected by IMEI/IMSI )

wdoekes, 2022-11-04:
Incomplete ticket.

CpServiceSpb, 2022-11-26:
Hi.
Completed by which commit ?

closed 2022-11-26 by wdoekes

CpServiceSpb, 2022-11-26:
Why do you not plant it ?
Huawei supports voice.

wdoekes, 2022-11-27:
You have not provided me with sufficient info to fix things.

I do not own (nor want to own) a dongle. I can merge PRs that look fine. And I can change things that are obvious.

"Almost the same issue as xyz" is not something I'm willing to spend time on.

CpServiceSpb, 2022-11-27:
Ok.

Which info do you need to perform it ?
I will collect and send it.

Regards ...

AdrianBaqueiro, 2023-05-19:
Same issue here

[dongle0]
exten=+xxxxxx;
imei=xxxxxx;
imsi=xxxxxxx;

not detected but E180 and E1552 work fine with the same config