wdoekes/asterisk-chan-dongle #143: E3372: timedout while waiting 'OK' in response to 'AT+CSQ'

mghadam, 2021-05-05:
[image]

I am getting this error with Huawei E3372s-153:

`
[2021-05-05 18:09:05] ERROR[8077]: chan_dongle.c:522 do_monitor_phone: [dongle4] timedout while waiting 'OK' in response to 'AT+CSQ'
`

So chan_dongle constantly resets the dongle and fails at this error. 

I see a proper response from AT+CSQ. However, the output of this command is not consistent and it occasionally contains RSSI / HCSQ lines like this:

```
$ echo AT+CSQ | atinout - /dev/ttyUSB14 -

^RSSI:26

^HCSQ:"LTE",60,52,106,24

+CSQ: 28,99

OK

$ echo AT+CSQ | atinout - /dev/ttyUSB14 -

+CSQ: 26,99

OK
```

I looked at chan_dongle.c:522 line and it is this code:

```
if (!at_wait (fd, &t))
{
    ast_mutex_lock (&pvt->lock);
    ecmd = at_queue_head_cmd (pvt);
    if(ecmd)
    {
        ast_log (LOG_ERROR, "[%s] timedout while waiting '%s' in response to '%s'\n", dev, at_res2str (ecmd->res), at_cmd2str (ecmd->cmd));
        goto e_cleanup;
    }
    at_enqueue_ping(&pvt->sys_chan);
    ast_mutex_unlock (&pvt->lock);
    continue;
}
```

Any ideas how can I debug this error?

shalzz, 2021-07-28:
Hi @mghadam 

This is an issue with E3372 where it CSQ timesout while the dongle is being initialized but works fine after.
You can try my fork here with few AT commands disabled for E3372 after which the dongle works fine (at least for sms)

mlundblad70, 2023-04-20:
Was this fixed?