wdoekes/asterisk-chan-dongle #148: E1550 broken by commit 3d2a97f26ba84473a0926a9fe9a32e451a56bc36

AlexHighTower, 2021-10-04:
Hello

Just updated to latest commit 3d2a97f26ba84473a0926a9fe9a32e451a56bc36 and all my E1550 now broken
They in state Dongle has connected, initializing... 
Before latest changes it was
Dongle has connected, initializing...
Dongle initialized and ready

now it stuck at command
write command 'AT+CMGR' expected response '+CMGR' length 12

looks like it sends AT+CMGR without any params and this is not supported

wdoekes, 2021-10-04:
Your subject says 3d2a97f26ba84473a0926a9fe9a32e451a56bc36 is the culprit. But I have a hard time believing that.

Are you using a big endian platform? Do things work if you revert that commit? Or is it perhaps a different commit.

Did you `git bisect` to find the cause between the last known good commit and the problematic commit?

Also, please add more verbose debug logs to your report. I don't own a dongle, I cannot trace what you're doing.

wdoekes, 2021-10-05:
I cannot parse what you're saying.

There are 13 commits from ca9b692 to 3d2a97f. Which commit is the oldest that fails? The one after ca9b692 (= 5a0ee5318) or 6073c91 or 3d2a97f ?

Please be precise.

If you say ca9b692 works. Then you could do:
```
git bisect
git bisect good ca9b692
git bisect bad 3d2a97f
```
and git will provide you with a binary search after which you can say `git bisect good` or `git bisect bad` according to whether the provided commit works or not.

wdoekes, 2021-10-05:
Having said that, I suspect that you need this:
```
--- a/at_command.c
+++ b/at_command.c
@@ -149,8 +149,8 @@ EXPORT_DEF int at_enqueue_initialization(struct cpvt *cpvt, at_cmd_t from_comman
                ATQ_CMD_DECLARE_STI(CMD_AT_CNUM, "AT+CNUM\r"),
                /* read the current voice mode, and return sampling
                 * rate, data bit, frame period */
-               ATQ_CMD_DECLARE_ST(CMD_AT_CVOICE, "AT^CVOICE?\r"),
-               ATQ_CMD_DECLARE_ST(CMD_AT_QPCMV, "AT+QPCMV?\r"), /* for Quectel */
+               ATQ_CMD_DECLARE_STI(CMD_AT_CVOICE, "AT^CVOICE?\r"),
+               ATQ_CMD_DECLARE_STI(CMD_AT_QPCMV, "AT+QPCMV?\r"), /* for Quectel */
 
                /* Get SMS Service center address */
                ATQ_CMD_DECLARE_ST(CMD_AT_CSCA, "AT+CSCA?\r"),
```

AlexHighTower, 2021-10-05:
> Having said that, I suspect that you need this:

it helped! thanks!

closed 2021-10-05 by wdoekes

rojer, 2022-01-15:
so, i just finished debugging this issue myself, came to submit a patch and found that it's already been fixed.

i think [these lines](https://github.com/wdoekes/asterisk-chan-dongle/blob/3d046f7d6842298c6838b5ce5b51d495d383b158/at_response.c#L377-L385) can also be removed, since `ATQ_CMD_DECLARE_STI` will plow ahead anyway, no need to manually restart the sequence.

commit 3d046f7d (2021-10-05): Do not choke on failed AT^CVOICE? or AT+QPCMV? during startup

ATQ_CMD_DECLARE_STI adds ATQ_CMD_FLAG_IGNORE so the error response on
either command does not abort the entire initialization task.

Without it, we migh get:

    [AT+QPCMV?]
    remove command 'AT+QPCMV' expected response 'OK' real 'ERROR' cmd
      16/22 flags 0x01 from queue
    remove task with 22 command(s) begin with 'AT' expected response
      'OK' from queue

where the failed AT+QPCMV? aborts the 22-item init task.

Closes #148, reported and tested by @AlexHighTower.