wdoekes/asterisk-chan-dongle #149: Quectel EC25

epipenau, 2021-10-16:
Hi,
I have attempted to use this module for the Quectel EC25 in a PinePhone.  Asterisk can Answer calls but I hear no audio.

I don't know where to go from here.

Any help would be greatly appreciated.

epipenau, 2021-11-03:
i changed the asterisk run as user to root, now when receiving an incoming call I get this:

    -- Executing [s@dongle-incoming:1] Answer("Dongle/dongle0-0400000000", "") in new stack
[Nov  3 11:56:20] ERROR[3876]: at_response.c:287 log_cmd_response_error: [dongle0] AT^DDSETEX (setup voice) failed
    -- Executing [s@dongle-incoming:2] Echo("Dongle/dongle0-0400000000", "") in new stack
[Nov  3 11:56:39] ERROR[3876]: at_response.c:1937 at_response: [dongle0] Receive NO CARRIER

I was not receving the AT^DDSETEX problem previously.  Maybe this sheds some light on the subject?

wdoekes, 2021-11-03:
One: if the dongle had replied positively to `AT+QPCMV?`, chan_dongle would not send `AT^DDSETEX`, but `AT+QPCMV=1,0`.

Two: I'm not sure Quectel wants anyone to use their stuff. I found this URL behind a paywall/loginwall:
https://www.quectel.com/download/quectel_gsm_audio_player_application_note_v3-2

PRs are welcome. But I cannot be of assistance without code snippets that _do_ work.

Here are some more people complaining that it doesn't work: https://github.com/bg111/asterisk-chan-dongle/issues/276 and https://forum.banana-pi.org/t/failed-to-cross-compile-asterisk-voice-channel-for-openwrt-solved/12558 and https://forums.quectel.com/t/ec25-at-qpcmd-not-supported/8196/2 . Here's a PDF 
http://www.szchonghan.com/English/upload/files/2017/8/Quectel_EC20%20R2.0_AT_Commands_Manual_V1.0.pdf which sports no reference to `QPCMV` at all, or this one https://sixfab.com/wp-content/uploads/2018/09/Quectel_EC25EC21_AT_Commands_Manual_V1.2.pdf with the same issue.

manojmat29, 2021-11-09:
Chan_quectel does not work and I've been modifying it and have got it to work for EC25. With few other modifications, it is also working with Sim7600 but that's not so stable. Many changes are required as the modified code was still using  Huawei specific URC for triggering many events. Will be publishing the code soon. And then it could be integrated with chan_dongle and also improved upon by a competent programmer like wdoekes.

epipenau, 2021-11-09:
Yeah nice.   I did a bit of research and found that most the AT commands were incorrect and not specific.  This is great to hear that you have done this manojmat29,  I can’t wait!

manojmat29, 2021-11-11:
Writeup and working code for Quectel here https://github.com/IchthysMaranatha/asterisk-chan-quectel
Simcom might take a few more days

epipenau, 2021-11-11:
Madness.  I’ll check it out today and let you know how it goes!  I have the PinePhone here and some WBT routers both with the EC25.

IchthysMaranatha, 2021-11-18:
Have added simcom support

wdoekes, 2022-01-17:
I did check this fork, but it does not seem to take merging the code back into chan_dongle into account.

If someone files a PR to merge code, I'll take a look. Closing for now.

closed 2022-01-17 by wdoekes

IchthysMaranatha, 2022-02-10:
@epipenau 
If you have not got it working on pinephone as yet, this might be of help https://www.jdieter.net/posts/2022/02/07/pinephone-call-quality/