wdoekes/asterisk-chan-dongle #151: Error parsing incoming message: Invalid charset

turbozapekanka, 2021-10-20:
I see this message in the log
SMS is not received.
I'm willing to provide more logs but not sure how and where to look

wdoekes, 2021-10-21:
Looks like you're hitting this bit:

https://github.com/wdoekes/asterisk-chan-dongle/blob/3d046f7d6842298c6838b5ce5b51d495d383b158/pdu.c#L825-L830

I assume you can enable debugging somehow. `core set debug 1` (up to 7) should give you more info in the debug log.

qtlin, 2021-12-22:
What about 
```
Error parsing incoming message '+CMGR: 0,,33
099123305323003002F4240DD069B7F93D6D4E010000122122919414000E7774595E0685E56550FE5DFF01' at possition 44: Can't parse OA
```
I am using https://github.com/bg111/asterisk-chan-dongle currently, is this fork capable of parsing the above message or it is the same?

miopa, 2022-01-04:
> SMS is not received.

It's received, but is stuck in the storage. You can delete them with `AT+CMGD=1,4` (it's not human readable anyway)

> is this fork capable of parsing the above message

It is.

wdoekes, 2022-01-17:
Thanks @miopa !

closed 2022-01-17 by wdoekes