wdoekes/asterisk-chan-dongle #153: SIP->GSM audio not working, only one side is heard

vogelfreiheit, 2021-12-17:
Hi,

I'm experiencing this issue:

`[2021-12-17 19:07:51] NOTICE[19243]: res_pjsip_sdp_rtp.c:150 rtp_check_timeout: Disconnecting channel 'PJSIP/888-00000002' for lack of audio RTP activity in 30 seconds`

Audio path from GSM->SIP works fine. But the other party cannot hear the audio from the SIP->GSM side. There is no firewall in-between, this is tested in the same LAN/VLAN. iptables empty. Using Zoiper on a device in the same network yields the same results.

I have enabled and disabled jitterbuffer and results don't change.

Dongle information:

```
-------------- Status -------------
  Device                  : dongle0
  State                   : Free
  Audio                   : /dev/ttyUSB1
  Data                    : /dev/ttyUSB2
  Voice                   : Yes
  SMS                     : Yes
  Manufacturer            : huawei
  Model                   : K3765
  Firmware                : 11.126.03.04.521
  IMEI                    : XXX
  IMSI                    : XXX
  GSM Registration Status : Registered, home network
  RSSI                    : 13, -87 dBm
  Mode                    : No Service
  Submode                 : No service
  Provider Name           : XXX
  Location area code      : XXX
  Cell ID                 : XXX
  Subscriber Number       : +XXX
  SMS Service Center      : +XXX
  Tasks in queue          : 0
  Commands in queue       : 0
  Call Waiting            : Disabled
  Current device state    : start
  Desired device state    : start
  When change state       : now
  Calls/Channels          : 0
    Active                : 0
    Held                  : 0
    Dialing               : 0
    Alerting              : 0
    Incoming              : 0
    Waiting               : 0
    Releasing             : 0
    Initializing          : 0
```

Sanitized log:

```
 Executing [XXX@from-internal:1] Macro("PJSIP/888-00000002", "user-callerid,LIMIT,EXTERNAL,") in new stack
    -- Executing [s@macro-user-callerid:1] Set("PJSIP/888-00000002", "TOUCH_MONITOR=1639768041.4") in new stack
    -- Executing [s@macro-user-callerid:2] Set("PJSIP/888-00000002", "CHANCONTEXT=") in new stack
    -- Executing [s@macro-user-callerid:3] Set("PJSIP/888-00000002", "CHANCONTEXT=") in new stack
    -- Executing [s@macro-user-callerid:4] Set("PJSIP/888-00000002", "CHANEXTENCONTEXT=888-00000002") in new stack
    -- Executing [s@macro-user-callerid:5] Set("PJSIP/888-00000002", "CHANEXTEN=888-00000002") in new stack
    -- Executing [s@macro-user-callerid:6] Set("PJSIP/888-00000002", "CALLERID(number)=501") in new stack
    -- Executing [s@macro-user-callerid:7] Set("PJSIP/888-00000002", "AMPUSER=501") in new stack
    -- Executing [s@macro-user-callerid:8] Set("PJSIP/888-00000002", "HOTDESCKCHAN=888-00000002") in new stack
    -- Executing [s@macro-user-callerid:9] Set("PJSIP/888-00000002", "HOTDESKEXTEN=501") in new stack
    -- Executing [s@macro-user-callerid:10] Set("PJSIP/888-00000002", "HOTDESKCALL=0") in new stack
    -- Executing [s@macro-user-callerid:11] ExecIf("PJSIP/888-00000002", "0?Set(HOTDESKCALL=1)") in new stack
    -- Executing [s@macro-user-callerid:12] ExecIf("PJSIP/888-00000002", "0?Set(CALLERID(name)=)") in new stack
    -- Executing [s@macro-user-callerid:13] GotoIf("PJSIP/888-00000002", "0?report") in new stack
    -- Executing [s@macro-user-callerid:14] ExecIf("PJSIP/888-00000002", "1?Set(REALCALLERIDNUM=501)") in new stack
    -- Executing [s@macro-user-callerid:15] Set("PJSIP/888-00000002", "AMPUSER=501") in new stack
[... 115 lines cut ...]
    -- Executing [s@macro-dialout-trunk:20] GotoIf("PJSIP/888-00000002", "0?bypass,1") in new stack
    -- Executing [s@macro-dialout-trunk:21] ExecIf("PJSIP/888-00000002", "1?Set(CONNECTEDLINE(num,i)=xxxx)") in new stack
    -- Executing [s@macro-dialout-trunk:22] ExecIf("PJSIP/888-00000002", "1?Set(CONNECTEDLINE(name,i)=CID:+XXX)") in new stack
    -- Executing [s@macro-dialout-trunk:23] ExecIf("PJSIP/888-00000002", "0?Set(CONNECTEDLINE(name,i)=CID:(Hidden)+XXX)") in new stack
    -- Executing [s@macro-dialout-trunk:24] GotoIf("PJSIP/888-00000002", "1?customtrunk") in new stack
    -- Goto (macro-dialout-trunk,s,30)
    -- Executing [s@macro-dialout-trunk:30] Set("PJSIP/888-00000002", "pre_num=AMP:dongle/dongle0/") in new stack
    -- Executing [s@macro-dialout-trunk:31] Set("PJSIP/888-00000002", "the_num=OUTNUM") in new stack
    -- Executing [s@macro-dialout-trunk:32] Set("PJSIP/888-00000002", "post_num=") in new stack
    -- Executing [s@macro-dialout-trunk:33] GotoIf("PJSIP/888-00000002", "1?outnum:skipoutnum") in new stack
    -- Goto (macro-dialout-trunk,s,34)
    -- Executing [s@macro-dialout-trunk:34] Set("PJSIP/888-00000002", "the_num=XXX") in new stack
    -- Executing [s@macro-dialout-trunk:35] Dial("PJSIP/888-00000002", "dongle/dongle0/XXX,300,T") in new stack
    -- Called dongle/dongle0/XXX
    -- Dongle/dongle0-0100000002 is making progress passing it to PJSIP/888-00000002
       > 0xb3e96a40 -- Strict RTP learning after remote address set to: XXX:59546
    -- Dongle/dongle0-0100000002 answered PJSIP/888-00000002
    -- Channel Dongle/dongle0-0100000002 joined 'simple_bridge' basic-bridge <546c01ea-57c4-4f9d-8095-badd5d8ca237>
    -- Channel PJSIP/888-00000002 joined 'simple_bridge' basic-bridge <546c01ea-57c4-4f9d-8095-badd5d8ca237>
[2021-12-17 19:07:51] NOTICE[19243]: res_pjsip_sdp_rtp.c:150 rtp_check_timeout: Disconnecting channel 'PJSIP/888-00000002' for lack of audio RTP activity in 30 seconds
    -- Channel PJSIP/888-00000002 left 'simple_bridge' basic-bridge <546c01ea-57c4-4f9d-8095-badd5d8ca237>
    -- Channel Dongle/dongle0-0100000002 left 'simple_bridge' basic-bridge <546c01ea-57c4-4f9d-8095-badd5d8ca237>
  == Spawn extension (macro-dialout-trunk, s, 35) exited non-zero on 'PJSIP/888-00000002' in macro 'dialout-trunk'
  == Spawn extension (from-internal, XXX, 10) exited non-zero on 'PJSIP/888-00000002'
```

vogelfreiheit, 2021-12-20:
Somehow it disappeared after disabling strict rport... I suspected a media channel transport issue, so it might have been old NAT states and what not. Keeping it open for a while in case it resurfaces.

wdoekes, 2022-01-17:
Looks like chan_pjsip config problem then. Good that you found it. Closing! :)

closed 2022-01-17 by wdoekes