wdoekes/asterisk-chan-dongle #163: GSM <->SIP Asterisk 18.14.0 FreeBSD 13.1 Huawei E169

VasOleMil, 2022-11-12:
No audio from GSM, modem is working well with Windows 10 and provider program on the same machine, audio present in both directions regardless of caller. Asterisk installed from ports, SIP<->SIP connection works fine. 

making sequence:

```
    ./bootstrap
    ./configure LDFLAGS="-L /usr/local/lib" --with-astversion=18.14.0 (command asterisk -V)
    add in Makefile block usage, line 43: CFLAGS = -g -O2 -O6 -Wno-out-of-line-declaration -fblocks ...
    add in the end makefile line LIBS = /usr/local/lib/libiconv.so
    gmake and copy your file to asterisk modules or use gmake install
```

```
CLI> dongle show device settings vodafone
------------- Settings ------------
  Device                  : vodafone
  Audio                   : /dev/cuaU0.1
  Data                    : /dev/cuaU0.2
  IMEI                    : 35963801387****
  IMSI                    : 25501515034****
  Channel Language        : en
  Context                 : vodafone-incoming
  Exten                   : +38095*******
  Group                   : 1
  RX gain                 : 0
  TX gain                 : 0
  U2Diag                  : 0
  Use CallingPres         : Yes
  Default CallingPres     : Presentation Allowed, Passed Screen
  Auto delete SMS         : Yes
  Disable SMS             : No
  Reset Dongle            : Yes
  SMS PDU                 : No
  Call Waiting            : auto
  DTMF                    : relax
  Minimal DTMF Gap        : 45
  Minimal DTMF Duration   : 80
  Minimal DTMF Interval   : 200
  Initial device state    : start
```
The only warning in console:
`WARNING[102589][C-00000002]: channel.c:1079 channel_indicate: [Dongle/vodafone-0100000001] Don't know how to indicate condition 22`

VasOleMil, 2022-11-12:
Updated modem firmware to the latest, still no audio at GSM->SIP:
```
CLI> dongle show devices
ID           Group State      RSSI Mode Submode Provider Name  Model      Firmware          IMEI             IMSI             Number        
vodafone     1     Free       16   5    4       Vodafone UA    E169       11.315.05.00.00   35963801387****  25501515034****  Unknown  
```

VasOleMil, 2022-11-13:
**CLI Debug** 
```
CLI> rtp set debug on
RTP Packet Debugging Enabled
CLI> core set verbose 10
Console verbose was 1 and is now 10.
```

**CLI GMS->SIP with hold** 
```
    -- Executing [+38095*******@vodafone-incoming:1] Dial("Dongle/vodafone-0100000000", "PJSIP/6000,60") in new stack
    -- Called PJSIP/6000
[Nov 13 02:24:50] WARNING[103721][C-00000001]: channel.c:1079 channel_indicate: [Dongle/vodafone-0100000000] Don't know how to indicate condition 22
    -- PJSIP/6000-00000000 is ringing
       > 0x804c55000 -- Strict RTP learning after remote address set to: 192.168.0.98:4010
    -- PJSIP/6000-00000000 answered Dongle/vodafone-0100000000
    -- Channel PJSIP/6000-00000000 joined 'simple_bridge' basic-bridge <4a608c57-f008-41d9-868c-7ac92f846b3b>
    -- Channel Dongle/vodafone-0100000000 joined 'simple_bridge' basic-bridge <4a608c57-f008-41d9-868c-7ac92f846b3b>
       > 0x804c55000 -- Strict RTP learning after remote address set to: 192.168.0.98:4010
    -- Started music on hold, class 'default', on channel 'Dongle/vodafone-0100000000'
       > 0x804c55000 -- Strict RTP learning after remote address set to: 192.168.0.98:4010
    -- Stopped music on hold on Dongle/vodafone-0100000000
Sent RTP packet to      192.168.0.98:4010 (type 00, seq 017366, ts 000160, len -000013)
...
Sent RTP packet to      192.168.0.98:4010 (type 00, seq 017656, ts 046560, len -000013)
    -- Channel PJSIP/6000-00000000 left 'simple_bridge' basic-bridge <4a608c57-f008-41d9-868c-7ac92f846b3b>
    -- Channel Dongle/vodafone-0100000000 left 'simple_bridge' basic-bridge <4a608c57-f008-41d9-868c-7ac92f846b3b>
  == Spawn extension (vodafone-incoming, +38095*******, 1) exited non-zero on 'Dongle/vodafone-0100000000'
```
**CLI GMS->SIP no hold** 
```
-- Executing [+38095*******@vodafone-incoming:1] Dial("Dongle/vodafone-0100000001", "PJSIP/6000,60") in new stack
    -- Called PJSIP/6000
[Nov 13 02:30:08] WARNING[103786][C-00000002]: channel.c:1079 channel_indicate: [Dongle/vodafone-0100000001] Don't know how to indicate condition 22
    -- PJSIP/6000-00000001 is ringing
       > 0x80a6e0000 -- Strict RTP learning after remote address set to: 192.168.0.98:4012
    -- PJSIP/6000-00000001 answered Dongle/vodafone-0100000001
    -- Channel PJSIP/6000-00000001 joined 'simple_bridge' basic-bridge <afa86dec-0822-4b0c-8b3e-82a0bdde4cba>
    -- Channel Dongle/vodafone-0100000001 joined 'simple_bridge' basic-bridge <afa86dec-0822-4b0c-8b3e-82a0bdde4cba>
    -- Channel PJSIP/6000-00000001 left 'simple_bridge' basic-bridge <afa86dec-0822-4b0c-8b3e-82a0bdde4cba>
    -- Channel Dongle/vodafone-0100000001 left 'simple_bridge' basic-bridge <afa86dec-0822-4b0c-8b3e-82a0bdde4cba>
  == Spawn extension (vodafone-incoming, +38095*******, 1) exited non-zero on 'Dongle/vodafone-0100000001'
```
**CLI SIP->GMS no hold** 
```
    -- Executing [38050*******@internal:1] Dial("PJSIP/6000-00000004", "Dongle/vodafone/+38050*******,240,tT") in new stack
    -- Called Dongle/vodafone/+38050*******
    -- Dongle/vodafone-0100000004 is making progress passing it to PJSIP/6000-00000004
       > 0x80367b000 -- Strict RTP learning after remote address set to: 192.168.0.98:4018
Sent RTP packet to      192.168.0.98:4018 (type 00, seq 003931, ts 000160, len -000013)
....
Sent RTP packet to      192.168.0.98:4018 (type 00, seq 004691, ts 121760, len -000013)
    -- Dongle/vodafone-0100000004 answered PJSIP/6000-00000004
    -- Channel Dongle/vodafone-0100000004 joined 'simple_bridge' basic-bridge <7c615bb8-961f-4a6f-89b6-ef0915589a84>
    -- Channel PJSIP/6000-00000004 joined 'simple_bridge' basic-bridge <7c615bb8-961f-4a6f-89b6-ef0915589a84>
Sent RTP packet to      192.168.0.98:4018 (type 00, seq 004692, ts 121920, len -000013)
...
Sent RTP packet to      192.168.0.98:4018 (type 00, seq 004808, ts 140480, len -000013)
    -- Channel PJSIP/6000-00000004 left 'simple_bridge' basic-bridge <7c615bb8-961f-4a6f-89b6-ef0915589a84>
  == Spawn extension (internal, 38050*******, 1) exited non-zero on 'PJSIP/6000-00000004'
    -- Channel Dongle/vodafone-0100000004 left 'simple_bridge' basic-bridge <7c615bb8-961f-4a6f-89b6-ef0915589a84>
```

VasOleMil, 2022-11-13:
Found reason, audio now works in both sides, after changing transport settings:

**pjsip.conf**

```
;===============TRANSPORT================================================================
[transport-udp-nat]
type=transport
protocol=udp
bind=0.0.0.0
local_net=192.168.0.0/24
local_net=127.0.0.0/32
external_media_address=176.36.XXX.XXX
external_signaling_address=176.36.XXX.XXX

;===============ENDPOINT TEMPLATES=======================================================

[endpoint-basic](!)
type=endpoint
context=internal
disallow=all
allow=ulaw
transport=transport-udp-nat
direct_media=no

[auth-userpass](!)
type=auth
auth_type=userpass

[aor-single-reg](!)
type=aor
max_contacts=1
remove_existing=yes
remove_unavailable=yes

;===============EXTENSION 6000

[6000](endpoint-basic)
auth=auth6000
aors=6000

[auth6000](auth-userpass)
password=6000
username=6000

[6000](aor-single-reg)

;===============EXTENSION 6001

[6001](endpoint-basic)
auth=auth6001
aors=6001

[auth6001](auth-userpass)
password=6001
username=6001

[6001](aor-single-reg)
```

closed 2022-11-13 by VasOleMil

wdoekes, 2022-11-14:
Thanks for clearing things up! :)

Could be that you have (Wifi) "client isolation/separation" on. Then direct traffic would be dropped.