wdoekes/asterisk-chan-dongle #164: Key press recognition

AlexanderZhirov, 2022-11-25:
I use the e1550 modem together with Asterisk 18.13.0. Incoming and outgoing calls and SMS work. How can I recognize keystrokes on my phone in Asterisk? Asterisk does not respond to keystrokes when I do it on my cell phone during a call.

AlexanderZhirov, 2022-11-26:
It was necessary to simply switch the e1550 device to modem mode.

wdoekes, 2022-11-27:
Cool. Thanks for reporting back.

AlexanderZhirov, 2022-11-27:
> Cool. Thanks for reporting back.

A small remark. Sometimes I have to physically reconnect the modem itself in the USB port. I took a note - first turn on the Asterisk - and then turn on the modem in the USB port. Otherwise, the modem loads faster and Asterisk behaves with it somehow wrong.

AlexanderZhirov, 2022-11-28:
@wdoekes Sometimes, when configuring files, rebooting Asterisk, modules, DTMF support disappears. It is necessary to pull it out of the USB and plug it in again, or reboot the system, since only after that the DTMF from the caller to the modem starts working. It would be very useful to try to identify the cause. Any ideas?

AlexanderZhirov, 2022-12-03:
Still, I think it's a problem with my Asterisk. I need time for testing. I will unsubscribe in the next few days.

AlexanderZhirov, 2022-12-06:
> Still, I think it's a problem with my Asterisk. I need time for testing. I will unsubscribe in the next few days.

Still checked and yes, the problem is, sometimes DTMF stops working when calling via dongle. After restarting the server, everything works fine. How would debug this whole thing? Any suggestions?

AlexanderZhirov, 2023-01-11:
To summarize a little:

Three weeks ago I built `Asterisk` for **Raspberry Pi 4B**. I also use two `E1550` modems. Throughout this time, there were no problems with `DTMF`. Most likely, on my previous PC there was a problem of USB forwarding to the virtual machine (I used an Asterisk machine in **VirtualBox** as a server), which contributed to failures. I think that was the problem.

[image]

closed 2023-01-11 by AlexanderZhirov