wdoekes/asterisk-chan-dongle #166: chan_dongle.so not compiled with the same compile-time options

lpcs007, 2023-01-10:
I'm trying to install the module on Asterisk 18.15.0 and I'm having the same error.
I tried 18.15.0 and 18.15.
the asterisk-dev package is already installed

I'm using VitalPBX 4

_Originally posted by @lpcs007 in https://github.com/wdoekes/asterisk-chan-dongle/issues/89#issuecomment-1376087927_

lpcs007, 2023-01-11:
> @lpcs007 Which distribution do you use?

VitalPBX 4 use debian 11

AlexanderZhirov, 2023-01-11:
> > @lpcs007 Which distribution do you use?
> 
> VitalPBX 4 use debian 11

Could you drop the compilation log here?

lpcs007, 2023-01-11:
> > > @lpcs007 Which distribution do you use?
> > 
> > 
> > VitalPBX 4 use debian 11
> 
> Could you drop the compilation log here?

`# git clone https://github.com/wdoekes/asterisk-chan-dongle.git
Cloning into 'asterisk-chan-dongle'...
remote: Enumerating objects: 901, done.
remote: Counting objects: 100% (83/83), done.
remote: Compressing objects: 100% (48/48), done.
remote: Total 901 (delta 37), reused 69 (delta 33), pack-reused 818
Receiving objects: 100% (901/901), 508.72 KiB | 2.61 MiB/s, done.
Resolving deltas: 100% (560/560), done.

# cd asterisk-chan-dongle/
# ./bootstrap
configure.ac:78: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:7: installing './install-sh'
configure.ac:7: installing './missing'

# ./configure --with-astversion=18.15.0
checking build system type... aarch64-unknown-linux-gnu
checking host system type... aarch64-unknown-linux-gnu
checking target system type... aarch64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking how to run the C preprocessor... gcc -E
checking for strip... strip
checking for rm... rm
checking for library containing iconv... none required
checking for sqlite3_open in -lsqlite3... yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
checking for fcntl.h... yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking sys/time.h usability... yes
checking sys/time.h presence... yes
[... 70 lines cut ...]
char_conv.c:98:10: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   98 |   if (p0 == -1 || p1 == -1) {
      |          ^~
char_conv.c:98:22: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   98 |   if (p0 == -1 || p1 == -1) {
      |                      ^~
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT cli.o -MF .cli.o.d -MP  -o cli.o -c cli.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT helpers.o -MF .helpers.o.d -MP  -o helpers.o -c helpers.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT manager.o -MF .manager.o.d -MP  -o manager.o -c manager.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT memmem.o -MF .memmem.o.d -MP  -o memmem.o -c memmem.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT ringbuffer.o -MF .ringbuffer.o.d -MP  -o ringbuffer.o -c ringbuffer.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT cpvt.o -MF .cpvt.o.d -MP  -o cpvt.o -c cpvt.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT dc_config.o -MF .dc_config.o.d -MP  -o dc_config.o -c dc_config.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT pdu.o -MF .pdu.o.d -MP  -o pdu.o -c pdu.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT mixbuffer.o -MF .mixbuffer.o.d  [... cut ...]  -MT smsdb.o -MF .smsdb.o.d -MP  -o smsdb.o -c smsdb.c
gcc  -shared -Xlinker -x -o chan_dongle.so app.o at_command.o at_parse.o at_queue.o at_read.o at_response.o chan_dongle.o channel.o char_conv.o cli.o helpers.o manager.o memmem.o ringbuffer.o cpvt.o dc_config.o pdu.o mixbuffer.o pdiscovery.o error.o smsdb.o -lsqlite3

# make install
Makefile:76: warning: ignoring prerequisites on suffix rule definition
strip chan_dongle.so
/usr/bin/install -c -m 644 chan_dongle.so /usr/lib/asterisk/modules
`

AlexanderZhirov, 2023-01-11:
@lpcs007  So, have you tried to check? Right here:

```sh
ls -l /usr/lib/asterisk/modules/chan_dongle.so
```

lpcs007, 2023-01-11:
> ```shell
> ls -l /usr/lib/asterisk/modules/chan_dongle.so
> ```

```
# ls -l /usr/lib/asterisk/modules/chan_dongle.so
-rw-r--r-- 1 root root 277904 Jan 11 14:55 /usr/lib/asterisk/modules/chan_dongle.so
```

AlexanderZhirov, 2023-01-11:
> > ```shell
> > ls -l /usr/lib/asterisk/modules/chan_dongle.so
> > ```
> 
> ```
> # ls -l /usr/lib/asterisk/modules/chan_dongle.so
> -rw-r--r-- 1 root root 277904 Jan 11 14:55 /usr/lib/asterisk/modules/chan_dongle.so
> ```

Everything is fine. You have built the module, you have it installed. You just need to activate it. Edit `/etc/asterisk/module.conf`:

```sh
[modules]
autoload=yes

load => chan_dongle.so
```

lpcs007, 2023-01-11:
> > > ```shell
> > > ls -l /usr/lib/asterisk/modules/chan_dongle.so
> > > ```
> > 
> > 
> > ```
> > # ls -l /usr/lib/asterisk/modules/chan_dongle.so
> > -rw-r--r-- 1 root root 277904 Jan 11 14:55 /usr/lib/asterisk/modules/chan_dongle.so
> > ```
> 
> Everything is fine. You have built the module, you have it installed. You just need to activate it. Edit `/etc/asterisk/modules.conf`:
> 
> ```shell
> [modules]
> autoload=yes
> 
> load => chan_dongle.so
> ```
> 
> and restart Asterisk.
> 
> After that, go to Asterisk and the `dongle` [commands](https://github.com/wdoekes/asterisk-chan-dongle#other-cli-commands) will work for you.

I followed your instruction but the module is still not loaded. I also restarted the server.
If I try to load the module manually I get the following result.

```
vitalpbx*CLI> module load chan_dongle.so
Unable to load module chan_dongle.so
Command 'module load chan_dongle.so' failed.
[2023-01-11 15:33:37] ERROR[2007]: loader.c:283 module_load_error: Module 'chan_dongle.so' was not compiled with the same compile-time options as this version of Asterisk.
[2023-01-11 15:33:37] ERROR[2007]: loader.c:283 module_load_error: Module 'chan_dongle.so' will not be initialized as it may cause instability.
[2023-01-11 15:33:37] ERROR[2007]: loader.c:283 module_load_error: Module 'chan_dongle.so' could not be loaded.
```

lpcs007, 2023-01-11:
> Everything is fine. You have built the module, you have it installed. You just need to activate it. Edit `/etc/asterisk/modules.conf`:
>
> ```shell
> [modules]
> autoload=yes
>
> load => chan_dongle.so
> ``` [...]

I followed your instruction and the module was not loaded. I also restarted the server.
If I try to load the module manually I get the following result:

```
vitalpbx*CLI> module load chan_dongle.so
Unable to load module chan_dongle.so
Command 'module load chan_dongle.so' failed.
[2023-01-11 15:33:37] ERROR[2007]: loader.c:283 module_load_error: Module 'chan_dongle.so' was not compiled with the same compile-time options as this version of Asterisk.
[2023-01-11 15:33:37] ERROR[2007]: loader.c:283 module_load_error: Module 'chan_dongle.so' will not be initialized as it may cause instability.
[2023-01-11 15:33:37] ERROR[2007]: loader.c:283 module_load_error: Module 'chan_dongle.so' could not be loaded.
```

AlexanderZhirov, 2023-01-11:
@lpcs007 The `chan_dongle.so` module will not be displayed in the CLI. Not all modules are displayed there, some need to be loaded in the `modules.conf` configuration file.

The first time I made a typo was when I wrote the post above. You need to make edits to the `moduleS.conf` file. Here is the right path: `/etc/asterisk/modules.conf`. I also have version 18.15. He should work for you.

lpcs007, 2023-01-11:
> @lpcs007 The `chan_dongle.so` module will not be displayed in the CLI. Not all modules are displayed there, some need to be loaded in the `modules.conf` configuration file.
>
> The first time I made a [...]

Well then. I already followed all the instructions you said and the module just doesn't load. I try to use the `dongle` command and it is unknown. Well then. I already followed all the instructions you said and the module just doesn't load. I try to use the 'dongle' command and it is unknown.

lpcs007, 2023-01-11:
> @lpcs007 hmmm... Please give me a link from where to download your distribution and write me the commands that you performed to install Asterisk and tools. I'll try to test it.

I followed this article https://vitalpbx.com/blog/install-vitalpbx-4-on-a-raspberry-pi-quick-guide/.
I didn't use any other commands than the one above to install the dongle. This is a fresh install of VitalPBX.

AlexanderZhirov, 2023-01-12:
@lpcs007 I figured out your problem. [Here you can read more details](https://itgala.xyz/module-was-not-compiled-with-the-same-compile-time-options-as-this-version-of-asterisk-2/).

What you need to do:

Find out the `AST_BUILDOPT_SUM` of any of the available modules that came with Asterisk, for example:

```sh
grep -Eao "[0-9abcdef]{32}" /usr/lib/asterisk/modules/chan_audiosocket.so
```

In the output you will get something like this:

```sh
da6642af068ee5e6490c5b1d2cc1d238
ac1bc5634224cfc998788edb90d34359
```

Try checking another module to identify a duplicate hash:

```sh
grep -Eao "[0-9abcdef]{32}" /usr/lib/asterisk/modules/chan_pjsip.so
```

```sh
da6642af068ee5e6490c5b1d2cc1d238
1e96606fb9870fa299df388df0a83a30
```

In my case, the hash matched `da6642af068ee5e6490c5b1d2cc1d238`. You may have another one.

Next, edit the header file `/usr/include/asterisk/buildopts.h` and change the value of the constant to the one you got above:

```c
...
#define AST_BUILDOPT_SUM "da6642af068ee5e6490c5b1d2cc1d238"
...
```

Then go to the source directory `chan_dongle` and execute:

```sh
./configure --with-astversion=18.15
make
make install
```

Next, you need to change the user and group of the installed module:

```sh
chown asterisk:asterisk /usr/lib/asterisk/modules/chan_dongle.so
```

Before loading the module directly, you need to create your own configuration file for third-party modules:

```sh
touch /etc/asterisk/vitalpbx/modules__40-mymodule.conf
```

Give the necessary rights:

```sh
chown asterisk:asterisk /etc/asterisk/vitalpbx/modules__40-mymodule.conf
```

Enter the data into the created file:

```sh
nano /etc/asterisk/vitalpbx/modules__40-mymodule.conf
```

```sh
[modules](+)
load => chan_dongle.so
```

And be sure to copy the [file](https://github.com/wdoekes/asterisk-chan-dongle/blob/master/etc/dongle.conf) `dongle.conf`, which is located in the dongle source directory in the directory `yourpathsource/etc/dongle.conf`, to the directory `/etc/asterisk`:

```sh
cp yourPathDongleSource/etc/dongle.conf /etc/asterisk
```

Also give the rights:

```sh
chown asterisk:asterisk /etc/asterisk/dongle.conf
```

Restart your Asterisk and check the functionality of the module:

```sh
systemctl restart asterisk.service
asterisk -rx "dongle show version"
```

I was given:

```sh
chan_dongle: Huawei 3G Dongle Channel Driver, Version 1.1, Revision 43gh=47cg
Project Home: https://github.com/bg111/asterisk-chan-dongle/
Bug Reporting: https://github.com/bg111/asterisk-chan-dongle/issues
```

Be careful and you will succeed! 😉

lpcs007, 2023-01-12:
@AlexanderZhirov now the module is running fine. You are the man!!!

Thanks a lot for the help!

AlexanderZhirov, 2023-01-12:
@lpcs007 I was glad to help 😉

closed 2023-01-12 by lpcs007