wdoekes/asterisk-chan-dongle #167: Build from latest master doesn't work

kosfango, 2023-02-01:
Hi All,

```
Connected to Asterisk 13.32.0 currently running on server (pid = 51368)
server*CLI> dongle show devices
ID           Group State      RSSI Mode Submode Provider Name  Model      Firmware          IMEI             IMSI             Number
datacard0    2     Free       9    0    0       Operator PR    E1550      11.608.12.00.143  <imei here>  <imsi here>  <my number here>
server*CLI> dongle ussd datacard0 *100#
[datacard0] Cannot encode GSM7
```
But is I revert git to commit:
git checkout -b cb610d4e5e7d7c7e75e9b8fdc549376a38b47544

All works fine. @wdoekes maybe my Asterisk is too depreceted?

Thanks

wdoekes, 2023-02-01:
Possibly caused by same issue as #126?

Can you get debug logs from both the non-working and the working versions?

I don't think your Asterisk is to blame.

kosfango, 2023-02-02:
Hi @wdoekes ,

Will it enough asterisk -rvvvd for debug?
For now, I don't sure that https://github.com/wdoekes/asterisk-chan-dongle/issues/126 linked with my issue.
I need to get debug logs before. 
BTW, my OS FreeBSD 12, libiconv from OS (not from ports collection)

Thanks

wdoekes, 2023-02-02:
> Will it enough asterisk -rvvvd for debug?

Don't know. Possibly without `-d` and then `core set debug 20 chan_dongle` instead.

We don't need the debug logs of other subsystems, I think.