wdoekes/asterisk-chan-dongle #178: Problems on E1550

peresmeshnikov, 2024-08-10:
Hello. I have a few problems with my E1550 (and E3131, E173 as well)
1. [dongle0] Received unexpected 'OK' every 30 sec. But looks like it's dont have any sense.
2. Incoming calls doesn't work correctly. I have dial extension for incoming calls
```
[incoming]

;exten => s,1,Dial(PJSIP/100)
exten => _+xxxxxxxxxxx,1,NoOp()
exten => _+xxxxxxxxxxx,2,Dial(PJSIP/100)
exten => _+xxxxxxxxxxx,n,Hangup()

exten => sms,1,System(echo '${STRFTIME(${EPOCH},,%Y-%m-%d %H:%M:%S)} — ${DONGLENAME} от ${CALLERID(num)} Текст: ${BASE64_DECODE(${SMS_BASE64})}' >> /var/log/asterisk/sms.txt)
exten => sms,n,Hangup()

```
when I call to my phone number which in usb modem I recieve a call on 100 internal number, but when I answer I hear nothing to both sides. And if I finish on internal number side on a caller side talk is still in progress. In the asterisk's console I see dongle have status "active 1".
3. Sometimes calls from internal number outside works well, sometimes it drops after a few seconds. Don't understand a reason. Probably signal problem or something else, but at least I know it can work.

Maybe problem in extensions? I am not familar with asterisk.
Asterisk 20.9.1, chan_dongle from this fork, Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux.
But on the ubuntu 24 x86 I have same problems.

ger-pakon, 2024-08-22:
How is your dongle definition? do you declared in [dongleX] section the number you used in the dialplan?

e.g. "exten=+49177XXXXXXX" for a german number

peresmeshnikov, 2024-08-24:
> How is your dongle definition? do you declared in [dongleX] section the number you used in the dialplan?
> 
> e.g. "exten=+49177XXXXXXX" for a german number

I had default value simething like "+12345678901".
I have changed to my region prefix but the problem is still here. I can get incomming call, I can hang up, but no voices. Looks like asterisk cannot make connection berween my mobile phone and internal asterisk client.